sta-titansrobotics/27294-IntoTheDeep #2: Build/Test/Revise 1

mrfabroa, 2024-10-01:


Walker21122008, 2024-10-07:
Objective to complete:
 - Finish the linear motion arm and ensure it fits within the frame. - Kailyn and Hasini
 - Finish the Drivetrain - Kai and Rio
 - prepare a method for intake - Thomas

Walker21122008, 2024-10-07:
**Hasini, Kailyn and Hamida** (Note for Kailyn: could you add the photo of the extender to this comment??)
- Started with building the linear arm extender and tested to see how low 2 extensions would extend up to.
**Progress to do:**
- Have to remove stuck screw with the help of Mr. Fabroa
- Have to make sure that the extensions are not surpassing the limits along with the drivetrain
      - A) If it surpasses, we will have to use a shorter extension
      - B)  If it doesn't surpass, we can continue with the progress.
**Parts used for Monday:**
- Extension 15 mm
- Slider plate
- M3x8mm low profile socket cap screw
- M3 nut x8
- M3 locknut x8
- Double-sided slider

Aletrioli, 2024-10-12:
Objectives: Thomas

- Build the starter bot intake to learn about the intake and how it should work
- Test attaching the intake to the parts team-members build to give an idea of how the design would function in the context of the larger robot
- Make modifications to resolve issues that arise or generally improve the design

closed 2024-10-16 by mrfabroa