sta-titansrobotics/27294-IntoTheDeep #10: Sprint 1 Review and Demo

mrfabroa, 2024-10-07:


Walker21122008, 2024-10-12:
[Uploading Hasini  & Kailyn - Arm Building Sprint Document.docx…]()

Aletrioli, 2024-10-12:
Successes, failures, and next steps:

Successes:
- The starter robot intake was build completely
- The design is able to reliably take in and spit out samples

Failures:
- The wheels were placed incorrectly, and consequently do not touch the floor, leaving the extrusions to drag along the floor
- Very unstable when attached to the slider
- The hex shaft extends far beyond the length of the intake

Next steps:
- Create a custom-ramp to replace the corrugated plastic
- Cut the hex shaft
- Move the wheels so they hold up the intake
- Look into shaving off some weight (possibly by replacing some extrusions with lighter, plastic printed pieces)

riochan26, 2024-10-15:
Successes:
* System was successfully brainstormed
* Motors are working as intended

Failures:
* The chain was extremely annoying to manipulate and use, leading to failure

Next Steps:
* Finish the chain and ensure it functions properly with the wheel systems
* Assemble the rest of the frame and ensure it is able to move freely and withstand the weight of the other systems
* Link up power to the motors and code the wheels to move properly according to gamepad input

closed 2024-10-16 by mrfabroa