sta-titansrobotics/27294-IntoTheDeep #11: Arm for reaching the lower and higher basket

Walker21122008, 2024-10-08:
**Hasini and Kailyn:** ((Note for Kailyn: could you add the photo of the extender to this comment??)
We finished two sets of the linear motion slider and checked how long they would extend. For one set of arms, we are using four linear sliders, which extend more than the higher basket, which is good. 
- We still have to connect the 2 separate sliders and connect them using surgical tubing. We also believe that we can attach the gear to the robot like the V2 class bot which would make the arm turn instead of making it go upright only. If it is slanted throughout the robot, it would help to make it fit within the frame. 
https://www.youtube.com/watch?v=AxRqJGVuRSE
- This video showcases the method that we are going to be using for now to ensure that the arm doesn't require the turning part.

kailyn-s, 2024-10-08:
October 8 - Assembled Two Sets of Linear Motion Sliders

[image]
[image]
[image]

kailyn-s, 2024-10-10:
**October 9, 2024 - Combining the Linear Motion Sliders**

Today we began with finishing off the motion sliders as well as attempting to connect them. By connecting them, we were able to reach a length that stretched from the ground to above the high basket, leaving more than enough room for our output method. We also added the pulleys which will further contribute to the function of the arm later on. The sliders are still a bit stiff and tends to get stuck at certain points when extending due to impurities in the extrusions which is an issue we plan to address tomorrow.

kailyn-s, 2024-10-10:
[image]
[image]

kailyn-s, 2024-10-11:
**October 10, 2024 Kailyn & Hasini**

Today we began with the base of the arm and utilizing the pulleys to produce the base motion of the sliders. As seen in the photos below, we threaded a string through the various pulleys to connect all the pieces together. By pulling on one end of the string, you are able to see how the arm can be extended to the height of the high basket.

kailyn-s, 2024-10-11:
[image]
[image]

Walker21122008, 2024-10-15:
**October 15, 2024 - Hasini and Kailyn**
Today, we were starting to make the second set of arms. so far we have 2 linear extenders and we are making fast progress now since we know the steps to build it.
- Mr. Fabroa gave us an idea for how we can use the pulley system since the pulley kit from the Go Bilda kit from the other team would be too small

**Issues:**
- We are still figuring out how to mount the motor.
- We have to figure out how to make the linear slides slide back down easily rather than getting stuck.

kailyn-s, 2024-10-16:
**October 15 - Creating a Second Linear Motion System**
[image]

closed 2024-10-16 by mrfabroa