sta-titansrobotics/27294-IntoTheDeep #16: Sprint 2 Review and Demo

mrfabroa, 2024-10-16:


Walker21122008, 2024-10-26:
[Uploading Hasini & Kailyn - Sprint 2_ Reaching for the High_Low Basket (1).docx…]()

Walker21122008, 2024-10-26:
[https://docs.google.com/document/d/1Rp3MuxtzS-E2kpX_Z_wKBnIBBkfcfQmH2LR4hlytWuE/edit?tab=t.0](url)

Walker21122008, 2024-10-26:
[Hasini & Kailyn - Sprint 2_ Reaching for the High_Low Basket (1).docx](https://github.com/user-attachments/files/17528710/Hasini.Kailyn.-.Sprint.2_.Reaching.for.the.High_Low.Basket.1.docx)

Aletrioli, 2024-10-28:
Successes, failures, and next steps:

Successes:
- The tower was completed and attached to the drive-train

Failures:
- I did not sketch out or plan designs for my intake as I spent more time than I would have thought assembling and incorporating the tower

Next steps:
- Sketch and then create a custom-ramp to replace the corrugated plastic
- Adjust the intake as necessary to function with the sliders
- Fully assemble the robot

riochan26, 2024-10-28:
Successes, Failures and Next Steps

Success:
Drivetrain is built without motors being hooked up
Integrated the Shoulder of the Arm into the the drivetrain

Failures:
None

Next Steps:
Code the drivetrain

closed 2024-10-31 by mrfabroa