sta-titansrobotics/27294-IntoTheDeep #17: Sprint 2 - Reaching for the High/Low Basket (Hasini & Kailyn)

kailyn-s, 2024-10-16:
(add goals for the timeframe)

kailyn-s, 2024-10-16:
**October 15, 2024 - Hasini and Kailyn**
Today, we were starting to make the second set of arms. so far we have 2 linear extenders and we are making fast progress now since we know the steps to build it.

Mr. Fabroa gave us an idea for how we can use the pulley system since the pulley kit from the Go Bilda kit from the other team would be too small
Issues:

We are still figuring out how to mount the motor.
We have to figure out how to make the linear slides slide back down easily rather than getting stuck.

kailyn-s, 2024-10-16:
October 15 - Creating a Second Linear Motion System
Screenshot 2024-10-15 at 9 38 07 PM

kailyn-s, 2024-10-16:
October 16, 2024 - Hasini and Kailyn

Today we discussed the possibility of mounting the motion system onto the robot. Many issues were brought up; however, the main obstacle we discovered was the dimension limits, in which we are likely exceeding vertically. The linear motion system is about 15 inches, and the drive train is about 2 inches (without the wheels). With these approximate dimensions, we  are already close to exceeding the limits. Furthermore, we recognized that only 3 extrusions were necessary in our build to reach the height of the high basket. Therefore, today, we spent the remainder of the time adjusting the extrusions to reduce the height and weight of our build in hopes for it to become beneficial in the long run. 

**Issues**

- exceeds the height limit
- heavy, may be difficult to mount at a horizontal/diagonal angle

kailyn-s, 2024-10-16:
**October 16 - Reducing the Height of the Linear Motion System**
[image]

Walker21122008, 2024-10-18:
**October 16, 2024 - Hasini and Kailyn**

Today, we discussed how to mount the motor on the robot. We tested the robot's motor at different angles, figured out which works best, and found some issues regarding the life. We also discussed how to tighten the gear more and how to secure the string more properly so that it wouldn't be removed.

**Issues:**

- String gets tangled
- Motor might be too heavy
- Has to be positioned properly

Walker21122008, 2024-10-21:
**October 18, 2024 - Hasini and Kailyn**
Started with making the second set of arms which consists of the claw to hang the specimens with Kailyn. After that, we checked how to attach the motor to the robot so it wouldn't be too faulty. We figured out that the pulleys which controlled the strings were a bit faulty since the black part often kept coming loose so, we tried using screws to test it out but realized that the pulleys were better.  We figured out that we had to loosen the linear sliders since we figured out that it was the one that made it hard for the linear sliders to slide. We tested out using both ends of the string to check if it would lift accordingly but, it often got tangled. While testing the motors out, the gears accidentally cut through my skin leading to a deep cut, so lesson learnt - not to mess with gears.

**Issues:**
- The string keeps winding up in the wrong position.
- The motor kept tilting the linear slider so we figured that it would be due to it standing upright. Once it is mounted slanted, it will help prevent the issue.
- Have to figure out how to pull the sliders backwards

Walker21122008, 2024-10-22:
October 18, 2024 - Hasini and Kailyn
We worked on tying the other end of the pulleys to the linear motion system so that we would be able to check if the sliders would slide back down. We continued working on the other set of arms to use as a backup set of arms to work as the claw.

Issues:
 - We have to complete the robot to be able to mount the arm completely.
 - Have to test out the linear motion system to see how it works

Walker21122008, 2024-10-23:
Today, we continued with building the arm for the claw since the other team members still had to mount the stand on the drivetrain to support the linear motion sliders. We discovered that some of the needed parts were missing like the spaceshaft so we decided to use two for now. We realized that the linear sliders would not be able to pull the robot up due to the string tension so we decided to use the other set of arms for the pull-up. We realized that we accidentally switched the motor to the opposite side so we had to undo the whole claw just to flip the motor onto the other side.

Walker21122008, 2024-10-24:
**October 23 2024 - Hasini and Kailyn**
Today, we continued with building the arm for the claw. We have finished the actual frame and have started to make the claw. We also had an idea to use another set of linear extenders to extend the intake/claw (Still choosing) to reach up to the highest basket. This would make use of multiple gears connected to the linear extenders to lift the specimen up to the high basket. We have made a set of linear extenders to attach to the claw. We believe that this would fit within the frame.

**Issues:**
- (TBA by Kailyn)

kailyn-s, 2024-10-25:
**October 24, 2024**

Today we began building the claw part that will pick up samples. As we followed along the instructions, we realized that the claw didn't have the ability to extend as far as we had hoped. With this, we began brainstorming ideas to utilize extrusions and other gears to attach the claw to a longer set of arms, therefore providing it with a much greater reach. 

**Issues:**
- we are missing a few pieces such as spacers and screws that are part of the starter bot

kailyn-s, 2024-10-25:
October 25, 2024
Today, building was slower due to the lack of people and hands available. Nonetheless, the assembly of the claw continued, just at a slower pace. Due to many mistakes, we began with fixing screws and pieces that were either reversed or in the wrong position due to a misunderstanding of the instructions. After, we were able to put together the claw to the arm and demonstrate how the gear worked to put the arm in motion

Issues:
- the gears don't line up
- some parts are loose, making the gears not as secure as they could be

[image]
[image]

closed 2024-10-31 by mrfabroa