sta-titansrobotics/27294-IntoTheDeep #22: Sprint 3 - Review and Demo

mrfabroa, 2024-10-31:


Walker21122008, 2024-11-10:
**Obstacles & Issues**
The intake was initially installed incorrectly, requiring multiple adjustments and re-installation to ensure it worked properly.
Long shaft extensions need to be cut to fit properly and ensure smooth operation of the arm mechanisms.
Continuous adjustments were required for the positioning of the arms to ensure they move correctly and don't interfere with each other.
Ensuring that different arms (intake and base) coordinate and operate smoothly together has been an ongoing challenge.
Verifying that the robot fits within height limits required additional adjustments, indicating potential design constraints.
**Success**
The arm was fitted successfully onto the robot and is lifting without any obstruction manually.  This fulfills the objective for this sprint.
The base arm was successfully coded for smooth movement and functionality. It was tested multiple times displaying its effectiveness.
I was able to program the specimen gripper planned for sprint 4 in sprint 3 itself.  The specimen gripper was tested and confirmed to work effectively, securely gripping the specimen.

**Failures**
I realized that I should have spent more time on the coding than on the robot adjustment realizing that the competition is coming within 14 days.

**Next steps**
Start to focus on the portfolio that I am done with my coding section and track down the team progress.
Once the codes of the other members are done, they will need to be integrated into a single program.

Aletrioli, 2024-11-12:
Thomas & Rio

**Obstacles & Issues:**
- The movement code mysteriously stopped working at some point, that issue must be resolved
- Everything was misaligned with everything else in our robot, so we had to disassemble and reassemble many chunks of the robot to allow the arm to be fully installed
- The elbow motor struggles to rotate against gravity, we will investigate the meshing and ratio of the gears

**Successes:**
- The robot was fully assembled
- A good amount of the tele-op code is already complete
- The movement worked (but then it didn't)

**Failures:**
- The wires for the arm were not managed
- The code for the arm was not completed
- Even though the movement initially worked, it broke (we are going to investigate why)

**Next steps:**
- Fix the movement code
- The wires for the arm should be managed
- The code for the arm should be done as to allow macro'd extension and retraction
- The autonomous code should be done
- Signage should be managed