sta-titansrobotics/27294-IntoTheDeep #25: Sprint 4:  Team Portfolio

mrfabroa, 2024-11-05:
PDF of Team Portfolio based on manual guidelines, with an emphasis on Think, Innovate and/or Control Award criteria.  Be sure review the [portfolio and award criteria  in the manual](https://ftc-resources.firstinspires.org/file/ftc/game/manual-06)

Define and assign the distribution of duties here and create the necessary sub issues. Be sure to set the necessary assignees and label (Portfolio) on related issues.

Walker21122008, 2024-11-06:
**Nov 5, 2024**
Today, I started off with a new document and started to make a cover page for it on Canva. Once I was done making a cover page for the contest's theme "Into the Deep", I started by adding the titles and subheadings required in a detailed structure. Here is a picture of the cover page.
[image]

Walker21122008, 2024-11-13:
Nov 5, 2024
Today, I updated the "Into the Deep" cover page to better align with its theme. I also began designing the "Who We Are" page, mapping out its visual layout. Here’s an image of the enhanced cover page.
[image]

Walker21122008, 2024-11-13:
Nov 8, 2024
Today, I started off with making the third page of the document: Making the legend and the timeline & planning section. I made use of some sea like elements to give it an into the deep vibe. I also took inspiration of some of the past portfolios of the 19447 team to make the legend and tweaked it a bit. Here is the legend and title design so far:
[image]

Walker21122008, 2024-11-13:
Nov 11, 2024
Today, I started by discussing the timeline and planning section. I then discussed how we used sprints and differentiated the timings for certain parts, such as building certain sections, coding, etc. Here is the sprint page.
[image]

Walker21122008, 2024-11-19:
Nov 13, 2024

Walker21122008, 2024-11-19:
Nov 14, 2024

Walker21122008, 2024-11-19:
Nov 15, 2024

Walker21122008, 2024-11-19:
Nov 18, 2024

Walker21122008, 2024-11-19:
Nov 19, 2024