sta-titansrobotics/27294-IntoTheDeep #30: Sprint 5 - Drivetrain modifications and coding (Kai and Rio)

Kaiwithsteak, 2024-11-29:


Kaiwithsteak, 2024-11-29:
Fixing/revamping drivetrain code.
- recoding movement controls for the left and right joystick
- adding right trigger on controller as a sprint button
- refine precision of drivetrain movement

riochan26, 2024-12-12:
Nov. 31th - Dec. 10th

As we finished recoding the teleop movement and refined the drivetrain, we were unable to continue autonomous, as the other two arm segments were being worked on. With nothing to do, Kai and I decided to disperse and help our group members finish their respective arm pieces and the robot, so we could finish the autonomous at a later date. Within this timespan, I incorporated several design changes to the robot, and helped Kailyn learn how to code the robot, while Kai was helping the other group members with their pieces.

riochan26, 2024-12-16:
Dec. 12th and Dec 13th

We redistributed the roles as we shifted our focus, I am now focusing on the intake and the code, while Kai is focusing on specimen clamp and will float around to help other people.