sta-titansrobotics/27294-IntoTheDeep #33: expansion hub

BOB-H-F, 2024-12-04:
The expansion hub does not seem to be working as no input is coming in from the device.

BOB-H-F, 2024-12-04:
Date: 11/29/2024

(what will be worked one): Research more about the expansion hub, Find solutions and talk with teammates about what has Been done to eliminate some processes.

(what worked one): I emailed The company that made the expansion hubs further explaining what we did and if there’s anything else that could be done to fix the We’ve been having. And I also did some research about what can be done and learned a bit about the expansion hub. Also also talked with teammates about it.

Any obstacles : almost every method I found to fix the issue already done and seem to not be effective

Picture/video: 
[image]

BOB-H-F, 2024-12-04:
Date: 12/03/2024

(what will be worked one):  today I’m going to take a look at if they emailed me back and if so see if the procedure that they have recommended may work if not do further research and continue testing with some few other things, if that doesn’t work emailed them to see if the expansion hub can be returned for a newer one, although very unlikely.

(what was worked one):  today they did email me back and I talked to the teacher and it seems that this may work but at first I didn’t think so, so I did a lot of research on it and where it connected and I connected the control hub to the battery to see what was going on and research what the different type of lights do and I found out that the control hub was probably having issues with Connecting to the battery fully as it had a blue light, which indicates Insignificant amount of energy, which was not the case, Because previously, there was a test done to see if that was the issue it was not.

Any obstacles : it took a lot of time researching, to try to pinpoint exactly why it was not working to fix the issue.
Picture/video: 
[image]

BOB-H-F, 2024-12-04:
Date: 12/04/2024

(what will be worked one):  today I’m going to get at least half of the instructions done from the email. 

(what was worked one): I got to the 11th part and was able to do most of the things in back of all the code before hand to which was great.

Any obstacles : it took a bit of time to actually do this, especially when the computers at school did not work to save the files that were needed to be saved so I had to borrow one of my teammates, computers to back up the code and also to connect the expansion hub, I was also needing advice for a part, but I wasn’t able to get it that day.

Picture/video:
[image]

Walker21122008, 2024-12-05:
(Hamida on Hasini's computer )

Date: 12/05/2024

(what will be worked one): finish what the email side:

(what was worked one):  I worked on the expansion hub today, I found out how to connect the computer to the control hub. and opined some files also watched a video on how to do it. 

Any obstacles: my team also needed the robot so my progress was stagnant, and I also was struggling with the files.

Picture/video:
[Uploading DriverAssitant_v4.91 (1).zip…]()

Walker21122008, 2024-12-06:
(Hamida on Hasini's computer )

Date: 12/06/2024

(what will be worked one): finish what the email side:

(what was worked one): I am almost finished , Mr fabroa help me find and locate the file I was struggling to fined last time, and now I just need to wait  for the robust to be free to use.

Any obstacles: my team also needed the robot so my progress was stagnant, 

Picture/video:

BOB-H-F, 2024-12-09:
Date: 12/06/2024

(what will be worked one): finish what the email side:

(what was worked one): I looked at the files and made sure everything was downloaded after that. I try to follow the 10th step on the program which was opening up the tools file and it didn’t quite work out very well because for some reason it wasn’t properly connecting to the control hub. so I got assistance from the teacher(mr. Fabora) in which I found out I need to use another device that may help work, but that also ended up being a little difficult because you needed to download a file that computer can’t really do.

Any obstacles: the computer wouldn’t properly connect to the control hub my teammates were also working on the robot so it was a bit difficult to be able to connect to the control hub and continue working. 

Picture/video:
[image]

BOB-H-F, 2024-12-10:
Date: 12/10/2024

(what will be worked one): finish what the email side and use expansion hub. 

(what was worked one):  again the teacher(mr. Fabora) was kind enough help me. And 
 everything that was said on the email was executed refreshing control hub and then also having to re-enter the configurations into the hub, but it didn’t work and even when trying to set up with different control hub it still didn’t work so I’m continuing my journey of trying to fix it tomorrow but today I also am writing an email to them (ftc)explaining what happened.

Any obstacles:  although all our hard work and trial, the expansion hub still did not work.

Picture/video:
[image]

BOB-H-F, 2024-12-13:
Date: 12/11/2024

(what will be worked on): helping  team members and hopefully getting a emailba bake from ftc.

(what was worked on): I helped my  team member to come up with an idea for what to do for the intake system next and also helped build part of it, as we decided to go back to the old intake system. 

Any obstacles: we dismantled the new intake  system and didn’t know what our teammate's opinion was on it  as they had not been there for a long time. 

Picture/video:

[image]

BOB-H-F, 2024-12-13:
Date: 12/12/2024

(what will be worked on):  reassigning roles, discussion and then plan out from there. 

(what was worked on):  we had a group meeting that would work in the future, and I sent a mother gmail to ftc for the expectation hub. 

Any obstacles:overall today was pretty good no obstacles

Picture/video:
[image]

BOB-H-F, 2024-12-13:
Date: 12/13/2024

(what will be worked one): 
Expansion hub, intake, discussion.

 
(what was worked one):
I talked to FTC about the expansion hub 

Any obstacles:NA

Picture/video:
[image]