sta-titansrobotics/27294-IntoTheDeep #35: Sprint 5 - Review and Demo

Walker21122008, 2024-12-14:


Walker21122008, 2024-12-14:
**Hasini and Thomas**

**Obstacles & Issues**
- Poor communication skills with one of our members in this objective due to sick leave and sometimes even wrong decisions
- Original intake idea (vertical intake) having to be scraped off the plan due to its ineffectiveness when taking samples from the samples chamber(it was ineffective since when it had to take blocks that are further away, the intake would not be able to take it due to the arm being at a weird angle).
- Had only 2 days left to recreate a new intake 

**Success**
- Finished making a intake
- Finished testing the code required for the intake(though it has not been specifically tested for the intake)
- Displayed **gracious professionalism** for sharing our green rubber wheels with team titan 5

**Failures**
- Not being able to finish testing the code on the intake due to not enough time there
- Not being able to achieve proper communication among our team members due to some members being absent(Sorry Mr. Fab 😭)

**Next steps**
- Start implement the code on the intake
- Make any small minor changes to the intake
- Start driving practices to be prepared for the competition
- Take the chance of using the "Activate" opportunity to improve our team communication skills :D

kailyn-s, 2024-12-18:
**Sprint 5 Review**

Over the course of this sprint, my main focus has been on completing prototypes that demonstrated the different possibilities we had to use as the specimen grabber. Since we previously realized our former specimen grabber was heavily flawed making it difficult to pick up specimens efficiently, we wanted to create a new build that allowed us to speed up the process of picking up and placing specimens. With this, I explored various options such as different materials and shapes that could potentially make it easier for drivers during this process. I came up with two main ideas: One involves the use of rubber wheels as grabber to significantly increase the friction between specimens and the grabbers, and one involves the use of a plastic claw to allow a simple pick up. The rubber wheels seemed better to hold the specimen as the friction made it less prone to falling out of the grabber; however, the plastic claw was better to pick up the specimen as it was able to open wider and had a smaller contact point, making it easier to contact with the specimen in a short amount of time. Next week, we plan to put back together the robot to test the different grabbers in terms of efficiency, accuracy, and proficiency. 

**Issues:**
- the plastic claw has very little grip on the specimen, meaning it is prone to being dropped, especially at high speeds
- the claw seems quite weak
- many parts are standing out as this is only a prototype and many lengths and measurements haven't been accounted for
- the gears tend to unmash often; during matches, they would have to be adjusted after each match

**Next Objectives:**
- put together the robot and rewire everything so that these two grabbers can be implemented and tested
- record data such as efficiency and productivity of both options to determine the more proficient option to use in competition
- with the chosen grabber, improve the build so that the parts don't impede the pick up process of specimens or other aspects (this may include finding shorter hex shafts, adding shaft collars to the ends to hold parts in place, replacing gear  and screw sizes, etc.)